steve-community/steve #1513: New install - 0.7.7 - update schema update failed -  Cannot change column 'chargeBoxId': used in a foreign key constraint

radmedics01, 2024-07-18:
#### Specifications
```
SteVe Version     :  3.6.0
Operating system  : Debian 12
JDK               : 17.0.11
Database          :  10.11.6-MariaDB
```

#### Expected Behavior

JAR should be generated with no issues / errors

#### Actual Behavior

After starting ./mvnw package the failed while executing V0_7_7__update.sql:
`[ERROR] Migration of schema `stevedb` to version "0.7.7 - update" failed! Please restore backups and roll back database and code!
...
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:7.15.0:migrate (default) on project steve: org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: Migration V0_7_7__update.sql failed
[ERROR] -----------------------------------
[ERROR] SQL State  : HY000
[ERROR] Error Code : 1833
[ERROR] Message    : Cannot change column 'chargeBoxId': used in a foreign key constraint 'stevedb/FK_chargeBoxId_c' of table 'stevedb/connector'
[ERROR] Location   : /opt/steve/steve-steve-3.6.0/src/main/resources/db/migration/V0_7_7__update.sql (/opt/steve/steve-steve-3.6.0/src/main/resources/db/migration/V0_7_7__update.sql)
[ERROR] Line       : 1
[ERROR] Statement  : ALTER TABLE `chargebox`
[ERROR] CHANGE COLUMN `chargeBoxId` `chargeBoxId` VARCHAR(255) NOT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `endpoint_address` `endpoint_address` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `ocppProtocol` `ocppProtocol` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointVendor` `chargePointVendor` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointModel` `chargePointModel` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargePointSerialNumber` `chargePointSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `chargeBoxSerialNumber` `chargeBoxSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `fwVersion` `fwVersion` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `fwUpdateStatus` `fwUpdateStatus` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `iccid` `iccid` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `imsi` `imsi` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `meterType` `meterType` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `meterSerialNumber` `meterSerialNumber` VARCHAR(255) NULL DEFAULT NULL COMMENT '' ,
[ERROR] CHANGE COLUMN `diagnosticsStatus` `diagnosticsStatus` VARCHAR(255) NULL DEFAULT NULL COMMENT ''`

#### Steps to Reproduce the Problem

  1. Execute ./mvnw package

...

radmedics01, 2024-07-18:
How can i fix that?

goekay, 2024-07-20:
it looks like you are using `10.11.6-MariaDB` which we do not support at the moment. this is something that we are aware of. mariadb started to drift from mysql with newer versions and compatibility started to become an issue. we could not look into it yet.

if you read readme carefully, you will see the mysql and mariadb versions we support.

radmedics01, 2024-07-20:
OK i didn't see it, now it's running, thank you very much

closed 2024-07-20 by radmedics01