steve-community/steve #1514: Transaction detail page should only show relevant meter values

goekay, 2024-07-20:
#### Checklist

- [x] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [x] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

Current impl of transaction detail page of a transaction shows every possible meter value a) for that transactionId or b) at that connector with the timestamp properties. That includes values such as temperature or voltage (that are irrelevant for this transactionId), and complicates the reading of the table.

#### Describe the solution you'd like

Include only relevant meter values filtered by
- unit equals `Wh` or `kWh`, or
- expected measurand `Energy.Active.Import.Register`

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

goekay, 2024-07-20:
we have [a pretty detailed helper method](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java#L39) that identifies energy-related meter values and handles a lot of edge cases.

for ex. if all fields (unit, measurand, etc.) are null, the ocpp spec interprets [this as energy value.](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java#L53-L57). this is possible to express in SQL but the outcome will be an even more complicated query that will be hard to maintain.

instead, i will select all rows where unit is one of the following: null, "", "Wh", "kWh". 

null and "" will be there to address backwards compatibility. then, i will filter all fetched rows through the helper method.

closed 2024-07-25 by goekay

rpluem, 2024-12-23:
Can this be made somewhat more flexible? Previously I could see the actual current in A and demand in W in the transaction details. This was valuable information for me, but I understand that people have different preferences. For me it would be enough if I could configure this via the `main-properties`, but configuring it via the Web UI might be preferred.

rpluem, 2024-12-23:
If it was clear to me what direction would be accepted (no change at all, configuration through `main.properties`, Web UI) I could probably create a PR.