steve-community/steve #1523: Connection Issues with MicroOcppSimulator and SteVe Using Docker

EasonChen11, 2024-07-30:
Hello,

I am a student currently learning about the Open Charge Point Protocol (OCPP) and am trying to set up a connection between MicroOcppSimulator and SteVe using Docker. Despite following the setup instructions, I am unable to establish a successful connection. Here are the detailed steps I've taken:

1. I have set up MicroOcppSimulator using Docker, and it starts without issues using the default settings.
2. I have similarly set up SteVe using Docker.
3. I created a Docker network to facilitate the connection between MicroOcppSimulator and SteVe.
4. In SteVe, I created a virtual ChargeBoxID, which I then configured in MicroOcppSimulator, along with the Backend URL `ws://172.18.0.3:8180/steve/websocket/CentralSystemService`.

Despite these steps, the connection isn't successful. Could you please advise if there are any additional settings or common pitfalls that I might be missing? Any guidance or shared experiences would be greatly appreciated.

Thank you!

jos-rk, 2024-07-31:
I've had similar issues getting a connection to the 3.7.0 version on docker.
I still haven't succeeded despite spending many hours adjusting my configuration and checking the logs.
3.6.0 still works almost out of the box, just git clone it and edit src/main/resources/config/docker/main.properties and docker-compose.yaml and launch it.

EasonChen11, 2024-07-31:
Thank you very much for your suggestion. I successfully built and established a connection with steve using your method. Your assistance has been invaluable. Thank you for your help!

goekay, 2024-08-07:
can you please post logs?

PxCeMobGP, 2024-08-14:
I have the same issue. When taking a look with WireShark, SteVe is answering my WebSocket request with http response 302
The redirection link (inside the 302 response) is "http://192.168.0.211:8180/steve/manager/signin"  which is the login page

Request Frame:
```
Hypertext Transfer Protocol
    GET /steve/websocket/CentralSystemService/PxC003 HTTP/1.1\r\n
    Host: 192.168.0.211:8180\r\n
    <Host: 192.168.0.211:8180\r\n>
    Upgrade: websocket\r\n
    <Upgrade: websocket\r\n>
    Connection: Upgrade\r\n
    <Connection: Upgrade\r\n>
    Sec-WebSocket-Key: x3JJHMbDL1EzLkh9GBhXDw==\r\n
    <Sec-WebSocket-Key: x3JJHMbDL1EzLkh9GBhXDw==\r\n>
    Sec-WebSocket-Protocol: ocpp1.6\r\n
    <Sec-WebSocket-Protocol: ocpp1.6\r\n>
    Sec-WebSocket-Version: 13\r\n
    <Sec-WebSocket-Version: 13\r\n>
    \r\n
    [Full request URI: http://192.168.0.211:8180/steve/websocket/CentralSystemService/PxC003]
    <Request: True>
    [HTTP request 19/19]
    [Prev request in frame: 8165]
    [Response in frame: 8612]
```
Response Frame:

```
Hypertext Transfer Protocol
    HTTP/1.1 302 Found\r\n
    Vary: Origin\r\n
    Vary: Access-Control-Request-Method\r\n
    Vary: Access-Control-Request-Headers\r\n
    Set-Cookie: JSESSIONID=node01exnyb66z9gjz1a6112air9kyr81.node0; Path=/steve\r\n
    <Set-Cookie: JSESSIONID=node01exnyb66z9gjz1a6112air9kyr81.node0; Path=/steve\r\n>
    Expires: Thu, 01 Jan 1970 00:00:00 GMT\r\n
    X-Content-Type-Options: nosniff\r\n
    X-XSS-Protection: 0\r\n
    X-Frame-Options: DENY\r\n
    Location: http://192.168.0.211:8180/steve/manager/signin\r\n
    <Location: http://192.168.0.211:8180/steve/manager/signin\r\n>
    Content-Length: 0\r\n
    <Content-Length: 0\r\n>
    \r\n
    <Response: True>
    [HTTP response 16/19]
    [Time since request: 0.011014000 seconds]
    [Prev request in frame: 6823]
    [Prev response in frame: 6828]
    [Request in frame: 7270]
    [Next request in frame: 7710]
    [Next response in frame: 7715]
    [Request URI: http://192.168.0.211:8180/steve/websocket/CentralSystemService/PxC003]

```

goekay, 2024-08-15:
releases 3.7.0 and 3.7.1 are working fine using the following:

```
curl \
    --http1.1 \
    --include \
    --no-buffer \
    --header "Connection: Upgrade" \
    --header "Upgrade: websocket" \
    --header "Host: localhost:8080" \
    --header "Origin: http://localhost:8080/steve/websocket/CentralSystemService/test-station-1" \
    --header "Sec-WebSocket-Key: x3JJHMbDL1EzLkh9GBhXDw==" \
    --header "Sec-WebSocket-Version: 13" \
    --header "Sec-WebSocket-Protocol: ocpp1.6" \
    http://localhost:8080/steve/websocket/CentralSystemService/test-station-1
```

however HEAD fails, and i get the following response

```
HTTP/1.1 302 Found
Vary: Accept-Encoding
Vary: Origin
Vary: Access-Control-Request-Method
Vary: Access-Control-Request-Headers
Set-Cookie: JSESSIONID=node0ab6qn0wqgxtp12lbyjg16woje0.node0; Path=/steve
Expires: Thu, 01 Jan 1970 00:00:00 GMT
X-Content-Type-Options: nosniff
X-XSS-Protection: 0
X-Frame-Options: DENY
Location: http://localhost:8080/steve/manager/signin
Content-Length: 0
```

probably, a regression was introduced with the big migration in https://github.com/steve-community/steve/pull/1519

closed 2024-08-15 by goekay

goekay, 2024-08-15:
hey all, thanks for the reports and making me aware of this issue. i merged the fix into the main branch. please test the HEAD of the repo, and tell me whether the fix resolves your issues.

PxCeMobGP, 2024-08-15:
Hi goekay,

thank you for the quick solution. Works perfectly!

EasonChen11, 2024-08-15:
Hello, thank you very much for your assistance. I encountered the following issue when trying to use the updated method:
DB log:
```yaml
2024-08-16 02:24:04 2024-08-15 18:24:04 0 [Note] mysqld: ready for connections.
2024-08-16 02:24:04 Version: '10.4.30-MariaDB-1:10.4.30+maria~ubu2004'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  mariadb.org binary distribution
2024-08-16 02:24:05 2024-08-15 18:24:05 8 [Warning] Aborted connection 8 to db: 'unconnected' user: 'unauthenticated' host: '172.28.0.3' (This connection closed normally without authentication)
2024-08-16 02:24:05 2024-08-15 18:24:05 9 [Warning] Aborted connection 9 to db: 'unconnected' user: 'unauthenticated' host: '172.28.0.3' (This connection closed normally without authentication)
2024-08-16 02:24:06 2024-08-15 18:24:06 10 [Warning] Aborted connection 10 to db: 'unconnected' user: 'unauthenticated' host: '172.28.0.3' (This connection closed normally without authentication)
2024-08-16 02:24:07 2024-08-15 18:24:07 11 [Warning] Aborted connection 11 to db: 'unconnected' user: 'unauthenticated' host: '172.28.0.3' (This connection closed normally without authentication)
2024-08-16 02:24:08 2024-08-15 18:24:08 12 [Warning] Aborted connection 12 to db: 'unconnected' user: 'unauthenticated' host: '172.28.0.3' (This connection closed normally without authentication)
2024-08-16 02:24:10 2024-08-15 18:24:10 13 [Warning] Aborted connection 13 to db: 'unconnected' user: 'unauthenticated' host: '172.28.0.3' (This connection closed normally without authentication)
2024-08-16 02:24:13 2024-08-15 18:24:13 14 [Warning] Aborted connection 14 to db: 'unconnected' user: 'unauthenticated' host: '172.28.0.3' (This connection closed normally without authentication)
```
App log:
```yaml
2024-08-16 02:23:56 2024/08/15 18:23:56 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:23:57 2024/08/15 18:23:57 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:23:58 2024/08/15 18:23:58 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:23:59 2024/08/15 18:23:59 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:24:00 2024/08/15 18:24:00 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:24:01 2024/08/15 18:24:01 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:24:02 2024/08/15 18:24:02 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:24:03 2024/08/15 18:24:03 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:24:04 2024/08/15 18:24:04 Waiting for tcp://mariadb:3306: dial tcp 172.28.0.2:3306: connect: connection refused.
2024-08-16 02:24:05 2024/08/15 18:24:05 Ready: tcp://mariadb:3306.
2024-08-16 02:24:05 /bin/sh: 1: ./mvnw: not found
2024-08-16 02:24:05 2024/08/15 18:24:05 Ready: tcp://mariadb:3306.
2024-08-16 02:24:05 /bin/sh: 1: ./mvnw: not found
2024-08-16 02:24:06 2024/08/15 18:24:06 Ready: tcp://mariadb:3306.
2024-08-16 02:24:06 /bin/sh: 1: ./mvnw: not found
```
I followed the manual to set up the Docker project and would like to ask if there might be an error in my setup. Thank you for your response.

EasonChen11, 2024-08-17:
DB log:
```yaml
2024-08-17 22:58:32 2024-08-17 14:58:32+00:00 [Note] [Entrypoint]: Entrypoint script for MariaDB Server 1:10.4.30+maria~ubu2004 started.
2024-08-17 22:58:32 2024-08-17 14:58:32+00:00 [Note] [Entrypoint]: Switching to dedicated user 'mysql'
2024-08-17 22:58:32 2024-08-17 14:58:32+00:00 [Note] [Entrypoint]: Entrypoint script for MariaDB Server 1:10.4.30+maria~ubu2004 started.
2024-08-17 22:58:32 2024-08-17 14:58:32+00:00 [Note] [Entrypoint]: Initializing database files
2024-08-17 22:58:35 
2024-08-17 22:58:35 
2024-08-17 22:58:35 PLEASE REMEMBER TO SET A PASSWORD FOR THE MariaDB root USER !
2024-08-17 22:58:35 To do so, start the server, then issue the following command:
2024-08-17 22:58:35 
2024-08-17 22:58:35 '/usr/bin/mysql_secure_installation'
2024-08-17 22:58:35 
2024-08-17 22:58:35 which will also give you the option of removing the test
2024-08-17 22:58:35 databases and anonymous user created by default.  This is
2024-08-17 22:58:35 strongly recommended for production servers.
2024-08-17 22:58:35 
2024-08-17 22:58:35 See the MariaDB Knowledgebase at http://mariadb.com/kb
2024-08-17 22:58:35 
2024-08-17 22:58:35 Please report any problems at http://mariadb.org/jira
2024-08-17 22:58:35 
2024-08-17 22:58:35 The latest information about MariaDB is available at http://mariadb.org/.
2024-08-17 22:58:35 
2024-08-17 22:58:35 Consider joining MariaDB's strong and vibrant community:
2024-08-17 22:58:35 https://mariadb.org/get-involved/
2024-08-17 22:58:35 
2024-08-17 22:58:35 2024-08-17 14:58:35+00:00 [Note] [Entrypoint]: Database files initialized
2024-08-17 22:58:35 2024-08-17 14:58:35+00:00 [Note] [Entrypoint]: Starting temporary server
2024-08-17 22:58:35 2024-08-17 14:58:35+00:00 [Note] [Entrypoint]: Waiting for server startup
2024-08-17 22:58:36 2024-08-17 14:58:36+00:00 [Note] [Entrypoint]: Temporary server started.
2024-08-17 22:58:39 2024-08-17 14:58:39+00:00 [Note] [Entrypoint]: GENERATED ROOT PASSWORD: 8LN"qJD`1%UPd:RW{V+TIEQNS;LE<KUe
2024-08-17 22:58:39 2024-08-17 14:58:39+00:00 [Note] [Entrypoint]: Creating database stevedb
2024-08-17 22:58:39 2024-08-17 14:58:39+00:00 [Note] [Entrypoint]: Creating user steve
2024-08-17 22:58:39 2024-08-17 14:58:39+00:00 [Note] [Entrypoint]: Giving user steve access to schema stevedb
2024-08-17 22:58:39 2024-08-17 14:58:39+00:00 [Note] [Entrypoint]: Securing system users (equivalent to running mysql_secure_installation)
2024-08-17 22:58:39 
2024-08-17 22:58:39 2024-08-17 14:58:39+00:00 [Note] [Entrypoint]: Stopping temporary server
2024-08-17 22:58:42 2024-08-17 14:58:42+00:00 [Note] [Entrypoint]: Temporary server stopped
2024-08-17 22:58:42 
2024-08-17 22:58:42 2024-08-17 14:58:42+00:00 [Note] [Entrypoint]: MariaDB init process done. Ready for start up.
2024-08-17 22:58:42 
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] Starting MariaDB 10.4.30-MariaDB-1:10.4.30+maria~ubu2004 source revision 928012a27ae2d1549f1b3e6975b9d22652bbea3a as process 117
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: Using Linux native AIO
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: Uses event mutexes
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: Number of pools: 1
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: Using SSE2 crc32 instructions
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] mysqld: O_TMPFILE is not supported on /tmp (disabling future attempts)
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: Initializing buffer pool, total size = 256M, instances = 1, chunk size = 128M
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: Completed initialization of buffer pool
2024-08-17 22:58:35 2024-08-17 14:58:35 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] InnoDB: 10.4.30 started; log sequence number 60943; transaction id 20
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] Plugin 'FEEDBACK' is disabled.
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Warning] 'user' entry 'root@fa137b5967b0' ignored in --skip-name-resolve mode.
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Warning] 'proxies_priv' entry '@% root@fa137b5967b0' ignored in --skip-name-resolve mode.
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] Reading of all Master_info entries succeeded
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] Added new Master_info '' to hash table
2024-08-17 22:58:36 2024-08-17 14:58:36 0 [Note] mysqld: ready for connections.
2024-08-17 22:58:36 Version: '10.4.30-MariaDB-1:10.4.30+maria~ubu2004'  socket: '/var/run/mysqld/mysqld.sock'  port: 0  mariadb.org binary distribution
2024-08-17 22:58:39 2024-08-17 14:58:39 0 [Note] mysqld (initiated by: unknown): Normal shutdown
[... cut ...]
2024-08-17 22:58:41 2024/08/17 14:58:41 Waiting for tcp://mariadb:3306: dial tcp 172.25.0.2:3306: connect: connection refused.
2024-08-17 22:58:42 2024/08/17 14:58:42 Waiting for tcp://mariadb:3306: dial tcp 172.25.0.2:3306: connect: connection refused.
2024-08-17 22:58:43 2024/08/17 14:58:43 Ready: tcp://mariadb:3306.
2024-08-17 22:58:43 /bin/sh: 1: ./mvnw: not found
```
I rebuilt using the code from #1319 and set up the project with ```docker compose up -d``` as described in the manual. The issue remains the same as before. However, I didn't encounter this problem when building with versions 3.7.0 and 3.6.