steve-community/steve #1524: Enhanced User Management and Functions for Steve

TSK7M, 2024-08-01:
### Describe the problem you are trying to solve

The current Steve project lacks a dedicated user web interface, which prevents users from accessing and managing their own charging data. Without this interface, users are unable to perform essential self-service functions such as viewing their charging status, managing their RFID tags, and controlling their charging sessions. This limitation results in an increased administrative workload and reduced user satisfaction, as users cannot independently manage their charging activities and must rely on administrators for even basic tasks.

### Describe the solution you'd like

1. **User Web Interface for Self-Service**
    - **Assign RFID Tags to Users:** Implement functionality that allows administrators to assign RFID tags to users. These tags will be linked to the user's account, enabling them to log into a dedicated user web interface.
    - **Access and Manage Charging Data:** Through the user web interface, users will be able to access and manage their own charging data. This includes:
        - **Viewing Charging Status:** Users can see the status of their current and past charging sessions, including details such as start time, end time, and energy consumed.
        - **Managing RFID Tags:** Users can view and manage their RFID tags, ensuring they are correctly assigned and active.
        - **Monitoring Current Sessions:** Users can monitor the status of their ongoing charging sessions in real-time.
    - **Self-Service Functions:** The user web interface will provide a range of self-service functions, allowing users to:
        - **Start and Stop Charging Sessions:** Users can remotely start and stop their charging sessions via the web interface, providing greater control and convenience.
        - **View Station Availability:** Users can check the availability of other charging stations, displayed as "available" or "occupied," helping them plan their charging activities.
        - **Reserve Charging Stations:** Users can reserve charging stations for specific times, ensuring they have access when needed.
    - **Customized Views:** The interface will be personalized, allowing users to see only their own data. This includes viewing their own charging history, managing their RFID tags, and setting personal charging limits. This personalized approach ensures security and relevance of the displayed information.

2. **Admin Web Interface for Management**
    - **Manage Permissions:** The admin web interface will enable administrators to manage user permissions, assign RFID tags to specific charging stations, and control who has access to which stations.
    - **Set Charging Limits and Rates:** Administrators can use the admin interface to set and manage charging limits (e.g., daily, weekly, monthly) and rates for different users and stations, ensuring efficient energy use and cost control.
    - **Monitor and Export Data:** The admin interface will provide tools for monitoring station usage, viewing detailed reports, and exporting data in CSV and PDF formats. Administrators can apply filters to the data before exporting, ensuring the reports are relevant and targeted.

### Describe alternative solutions or features you've considered

1. **Centralized Management without User Interface**
    - In this approach, all user data and permissions would be managed centrally by administrators, without providing users with a dedicated web interface.
    - **Limitations:** This method would increase the administrative workload as administrators would need to handle all user requests and data management tasks manually. It would also limit user autonomy and satisfaction as they would need to rely on administrators for all their needs.

2. **Basic User Notifications**
    - Users could receive notifications about their charging status and session details via email or SMS without having direct access to a web interface.
    - **Limitations:** This solution provides limited interaction and control for users. They would not be able to manage their sessions or view detailed charging data independently, leading to a less efficient and satisfying experience.

### Additional context

Introducing a dedicated user web interface will empower users to manage their own charging activities, leading to a more efficient and satisfying experience. This will reduce the administrative burden by enabling self-service functionalities and provide a more structured and secure way to handle user-specific data and permissions. 

By allowing users to interact directly with the system, overall operational efficiency and user satisfaction will be significantly enhanced. Users will benefit from greater control over their charging sessions, real-time access to their data, and the ability to make reservations and manage their RFID tags independently. This personalized and user-centric approach will also foster a sense of ownership and engagement among users, contributing to the successful adoption and use of the Steve project’s charging infrastructure.

TSK7M, 2024-08-01:
### Feature Request: Extended Features for Charging Rates and Cost Export

Dear Steve Development Team,

I would like to propose a set of features that would significantly enhance the user management and cost tracking functionalities of the Steve system. These enhancements aim to improve the flexibility and accuracy of charging rate management, cost calculation, and export capabilities. Here are the detailed feature requests:

#### Charging Rates per Charging Station/Wallbox

**Defining Charging Rates**
- **Define Different Rates:** Implement functionality to define different charging rates per charging station or wallbox.
- **Administrative Interface:** Create an administrative interface that allows defining and managing rates based on various factors such as time of day, day of the week, or energy consumption.

**Integration of Charging Rates into the System**
- **Backend Logic:** Adjust the backend logic to account for the defined rates during the use of charging stations.
- **Dynamic Adjustment:** Ensure that the rates can be dynamically adjusted and correctly applied when using the stations.

#### Cost Calculation and Display

**Calculating Costs per User**
- **Real-Time Calculation:** Implement a function to calculate costs based on the defined rates and the usage of each charging station or wallbox.
- **Accurate Billing:** Ensure that costs are calculated in real-time or at regular intervals for accurate billing.

**Displaying Costs in the Web Interface**
- **User Cost Display:** Integrate a display that shows users their costs based on usage and the defined rates.
- **Clear and Understandable:** Ensure that costs are displayed clearly and understandably in the user interface.

#### Extension of Export Functions

**Exporting Costs per User in CSV and PDF**
- **Extended Export Functionality:** Extend the existing export function to include costs per user and day in CSV and PDF exports.
- **Data Filters:** Implement filters that allow sorting and adjusting data before export, such as by period, user, or charging station.

**Automatic Billing**
- **Automated Bill Generation:** Implement a function that automatically generates bills and exports them in CSV or PDF format.
- **Comprehensive Billing Information:** Ensure that the bills contain all necessary information for direct processing.

#### Flexible Charging Rates per User and Day/Time

**Defining Flexible Charging Rates**
- **User-Specific Rates:** Allow charging rates to be assigned per user, per day/time, and per charging station or wallbox.
- **Flexible Rate Management:** Develop a flexible rate management system where administrators can assign rates based on various combinations, such as:
  - Global rates for all users and all stations.
  - User-specific rates for all their tags and stations.
  - Day-specific rates for all users.
  - Combination rates (e.g., specific user on a specific day at a specific station).

**Integration of Flexible Rates**
- **Backend Support:** Ensure the backend logic can handle and apply these flexible rate combinations during charging sessions.
- **Administrative Configuration:** Update the administrative interface to support the detailed configuration of these flexible rates.

I believe these enhancements would greatly improve the user experience and administrative control within the Steve system.

Thank you.

TSK7M, 2024-08-01:
### Benefits and Impact of Proposed Features

**Enhanced User Experience**

1. **Personalized Views and Permissions:**
   - **Intuitive Interface:** By providing personalized views and permissions, the user interface becomes more intuitive and organized. Each user sees only the information relevant to them, such as the status of their own RFID tags and reserved charging stations. This reduces information overload and simplifies navigation.
   
2. **Individual Account Access and Reservation System:**
   - **User Accounts:** Users have individual accounts that allow them to log into the web interface. Through their account, they can access, monitor, and export their specific data as needed.
   - **Reservation System:** Users can book charging stations using an integrated reservation system. A calendar visually displays which time slots are booked or available, enabling efficient planning and management of their charging times.

**Increased Control and Security**

1. **Permission Management:**
   - **Detailed Permissions:** Detailed permission management ensures that only authorized users can access specific charging stations. This prevents unauthorized access and misuse of the charging infrastructure.
   - **Station Assignment:** Each charging station can be assigned to specific RFID tags, so only authorized users can use the station.

2. **Charging Limits for Cost Control:**
   - **Energy Monitoring:** Implementing charging limits (e.g., daily, weekly, or monthly) for RFID tags helps monitor energy consumption and control costs.
   - **Budget Management:** This ensures that usage stays within set budgets, and users can plan and adjust their charging activities to stay within the prescribed limits.

**Operational Efficiency**

1. **Reservation System and Calendar:**
   - **Advance Booking:** An integrated reservation system allows users to book charging stations for specific time periods in advance.
   - **Visual Calendar:** A visual calendar clearly shows booked and available time slots, allowing users to plan their charging times efficiently. This ensures optimal use of the charging infrastructure and minimizes conflicts in station usage.

2. **Optimization of Station Availability:**
   - **Avoid Bottlenecks:** Targeted reservation and scheduling help avoid bottlenecks and optimize station utilization.
   - **Increased Satisfaction:** This reduces wait times and increases overall user satisfaction by ensuring that charging stations are available when needed.

**Enhanced Reporting**

1. **Improved Export Functions:**
   - **CSV and PDF Export:** The ability to export data in both CSV and PDF formats significantly enhances reporting capabilities. Users can export their charging activities and other relevant data in the format that best suits their needs.

2. **Filtering Before Export:**
   - **Targeted Reports:** Users can apply specific filters before exporting data to sort and refine it according to certain criteria. This allows for targeted and detailed reports tailored to the needs of users or management. Enhanced reports support more precise data analysis and help in making informed decisions.

3. **Better Data Analysis:**
   - **Deeper Insights:** Combining filtered data with flexible export options provides deeper insights. Companies can identify trends and patterns, plan charging infrastructure more efficiently, and make strategic decisions based on detailed analyses.

These features provide comprehensive improvements in user experience, control, efficiency, and data analysis, significantly enhancing the management and utilization of the charging infrastructure within the Steve system.

TSK7M, 2024-08-03:
NRGkick GPS Data Integration for Tariff and Charging Location Restrictions

The NRGkick device sends GPS data via OCPP, allowing for local tariff and charging availability restrictions. When an NRGkick device is connected using a specific TAG/RFID, it would be highly beneficial to link the GPS position with the RFID tag. This would enable location-based restrictions, ensuring that users can only charge at designated locations and apply tariffs based on their GPS positions. This feature is crucial for mobile charging, as it allows for the enforcement of location-specific charging policies and tariff management.

juherr, 2024-08-03:
It looks you are asking for many features in the same issue.
Maybe you should dispatch them in one issue by feature request in order to clarify.

About user management, there is currently no user in steve, just a data named user.
Notice that Steve is a dedicated app for operators, not for end-users.

End-users application will be different between any companies (price logic, ui, features, ...) and can be a totally standalone application.
First, you need to define the data flow of your system (by api, by directly reading the database, ...).

TSK7M, 2024-08-03:
_Should I package the topics listed below into separate requests?_

### Project: Enhanced User Management and Functions for Steve

To ensure clarity and effective tracking, the topics have been divided into separate tasks. Each task is outlined below with its subtasks.

#### Main Objective
Improve the user and administrative functions of Steve by introducing a user-centric web interface and enhanced administrative features.

#### 1. User Web Interface for Self-Service

1.1 **Assign RFID Tags to Users**
   - **Task**: Implement functionality allowing administrators to assign RFID tags to users.
   - **Description**: RFID tags will be linked to user accounts, enabling users to log into a dedicated web interface.

1.2 **Access and Manage Charging Data**
   - **Task**: Implement a web interface allowing users to access and manage their charging data.
     - **Subtasks**:
       - View charging status
       - Manage RFID tags
       - Monitor current charging sessions in real-time

1.3 **Self-Service Functions**
   - **Task**: Implement various self-service functions in the user interface.
     - **Subtasks**:
       - Start and stop charging sessions
       - View station availability
       - Reserve charging stations
       - Personalized views to show user-specific data

#### 2. Admin Web Interface for Management

2.1 **Manage Permissions**
   - **Task**: Develop an admin web interface for managing user permissions.
     - **Subtasks**:
       - Assign RFID tags to specific charging stations
       - Control access to charging stations

2.2 **Set Charging Limits and Rates**
   - **Task**: Implement functions for setting and managing charging limits and rates.
     - **Subtasks**:
       - Set daily, weekly, monthly limits
       - Manage user-specific and station-specific rates

2.3 **Monitor and Export Data**
   - **Task**: Develop tools for monitoring station usage and exporting data.
     - **Subtasks**:
       - View detailed reports
       - Export data in CSV and PDF formats
       - Apply filters before exporting

#### 3. Enhanced Features for Charging Rates and Cost Export

3.1 **Define Charging Rates**
   - **Task**: Implement functionality to define different charging rates per charging station.
     - **Subtasks**:
       - Administrative interface to define and manage rates
       - Adjust backend logic to apply defined rates during station use

3.2 **Cost Calculation and Display**
   - **Task**: Implement a function to calculate costs based on defined rates.
     - **Subtasks**:
       - Real-time cost calculation
       - Display costs in the web interface

3.3 **Extend Export Functions**
   - **Task**: Extend the existing export functions to include costs per user in CSV and PDF formats.
     - **Subtasks**:
       - Implement filters to adjust data before export
       - Automatic bill generation and export in CSV or PDF
[... 58 lines cut ...]
  - Export data in CSV and PDF formats
  - Apply filters before exporting

#### 3. Enhanced Features for Charging Rates and Cost Export
- **Ticket 7**: Define Charging Rates
  - Administrative interface to define and manage rates
  - Adjust backend logic to apply defined rates during station use
- **Ticket 8**: Cost Calculation and Display
  - Real-time cost calculation
  - Display costs in the web interface
- **Ticket 9**: Extend Export Functions
  - Implement filters to adjust data before export
  - Automatic bill generation and export in CSV or PDF
- **Ticket 10**: Flexible Charging Rates per User and Day/Time
  - Define user, day, and time-based rates
  - Backend support for flexible rate combinations

#### 4. NRGkick GPS Data Integration for Tariff and Charging Location Restrictions
- **Ticket 11**: Integrate GPS Data
  - Link GPS position with RFID tag
  - Enforce location-based charging restrictions and tariff management

By organizing the tasks in this manner, you can track progress more effectively and ensure that each component is developed and tested thoroughly.

---

juherr, 2024-08-05:
Most of the features listed here are end-user/eMSP-oriented: you should start a new project for them.
Then, Steve will be one of your CPO stacks and you'll just have to determine how you want to deal with it in terms of data and notifications.

About billing, you can either integrate solutions like https://github.com/killbill/killbill or https://github.com/getlago/lago

closed 2024-08-05 by TSK7M

juherr, 2024-08-05:
Thanks ChatGPT! 😅

goekay, 2024-08-05:
many of the things you shared in your last comments are possible with steve.

what is possible:
- Support for OCPP 1.6J
- Management UI (in english)
- Multiple connectors
- Data export as CSV
- Cross-Platform and Cloud-Ready
- Intuitive User Interface (well... as good as it gets with my skillset)
- Flexible Database Support: DB migration scripts are already there for setup
- Enhanced Management Functions (very vague term but yet, an admin have full control on UI)

what is not possible:
- Support for OCPP 2.0
- Management UI (in german)
- Flexible Database Support: only support mysql and mariadb

TSK7M, 2024-08-05:
@juherr 
Whether the answer is really helpful now!?
So, if you think the requirements come from ChatGPT... Okay...
I understand that you have no interest in a better web interface.
You have an application area and that's it. Nothing more.

As I said, I'll try my luck elsewhere.
But thanks for the great answer.

@goekay At least someone here is serious. Thanks, but @juherr doesn't care, it's all just ChatGPT.

TSK7M, 2024-08-05:
Thank you, Mr. @juherr, for your arrogance. It is really nice here; I like coming here. People are taken seriously. It is really nice here with the "German" developers.

goekay, 2024-08-05:
please.

@juherr was responding constructively until your last message. he is one of our valued contributors. he cares.

i could also have criticised your post... because you apparently did not even try out steve, and use it and experiment with it to see what the missing features are. instead of a more useful and to-the-point issue description, you dropped one wall-of-text after another with the undertone of "here are my 1 million feature requests". and these texts and messages were very verbose. they were not compact and information-dense. 

these kinds of issues therefore put more mental load on us, contributors, who do this for free among ten thousand other responsibilities that come with life.

juherr, 2024-08-05:
Thanks @goekay 

That's really strange that the account is deleted now 🤔 

@ TS (I didn't save your full name) Sorry, it was not my intention to hurt you, but your messages sounded a bit generated (and I even don't know if it is good or bad thing).

BTW, for the record:

-----

### Describe the problem you are trying to solve
The current Steve project lacks a dedicated user web interface, which prevents users from accessing and managing their own charging data. Without this interface, users are unable to perform essential self-service functions such as viewing their charging status, managing their RFID tags, and controlling their charging sessions. This limitation results in an increased administrative workload and reduced user satisfaction, as users cannot independently manage their charging activities and must rely on administrators for even basic tasks.

### Describe the solution you'd like
1. User Web Interface for Self-Service

- Assign RFID Tags to Users: Implement functionality that allows administrators to assign RFID tags to users. These tags will be linked to the user's account, enabling them to log into a dedicated user web interface.
- Access and Manage Charging Data: Through the user web interface, users will be able to access and manage their own charging data. This includes:
  - Viewing Charging Status: Users can see the status of their current and past charging sessions, including details such as start time, end time, and energy consumed.
  - Managing RFID Tags: Users can view and manage their RFID tags, ensuring they are correctly assigned and active.
  - Monitoring Current Sessions: Users can monitor the status of their ongoing charging sessions in real-time.
- Self-Service Functions: The user web interface will provide a range of self-service functions, allowing users to:
  - Start and Stop Charging Sessions: Users can remotely start and stop their charging sessions via the web interface, providing greater control and convenience.
  - View Station Availability: Users can check the availability of other charging stations, displayed as "available" or "occupied," helping them plan their charging activities.
  - Reserve Charging Stations: Users can reserve charging stations for specific times, ensuring they have access when needed.

- Customized Views: The interface will be personalized, allowing users to see only their own data. This includes viewing their own charging history, managing their RFID tags, and setting personal charging limits. This personalized approach ensures security and relevance of the displayed information.

2. Admin Web Interface for Management

- Manage Permissions: The admin web interface will enable administrators to manage user permissions, assign RFID tags to specific charging stations, and control who has access to which stations.
- Set Charging Limits and Rates: Administrators can use the admin interface to set and manage charging limits (e.g., daily, weekly, monthly) and rates for different users and stations, ensuring efficient energy use and cost control.
- Monitor and Export Data: The admin interface will provide tools for monitoring station usage, viewing detailed reports, and exporting data in CSV and PDF formats. Administrators can apply filters to the data before exporting, ensuring the reports are relevant and targeted.

### Describe alternative solutions or features you've considered

1. Centralized Management without User Interface
- In this approach, all user data and permissions would be managed centrally by administrators, without providing users with a dedicated web interface.
- Limitations: This method would increase the administrative workload as administrators would need to handle all user requests and data management tasks manually. It would also limit user autonomy and satisfaction as they would need to rely on administrators for all their needs.

2. Basic User Notifications

- Users could receive notifications about their charging status and session details via email or SMS without having direct access to a web interface.
- Limitations: This solution provides limited interaction and control for users. They would not be able to manage their sessions or view detailed charging data independently, leading to a less efficient and satisfying experience.

### Additional context
Introducing a dedicated user web interface will empower users to manage their own charging activities, leading to a more efficient and satisfying experience. This will reduce the administrative burden by enabling self-service functionalities and provide a more structured and secure way to handle user-specific data and permissions.

By allowing users to interact directly with the system, overall operational efficiency and user satisfaction will be significantly enhanced. Users will benefit from greater control over their charging sessions, real-time access to their data, and the ability to make reservations and manage their RFID tags independently. This personalized and user-centric approach will also foster a sense of ownership and engagement among users, contributing to the successful adoption and use of the Steve project’s charging infrastructure.

-----

### Feature Request: Extended Features for Charging Rates and Cost Export

Dear Steve Development Team,

I would like to propose a set of features that would significantly enhance the user management and cost tracking functionalities of the Steve system. These enhancements aim to improve the flexibility and accura [... cut ...]
- Intuitive User Interface: The web-UI is user-friendly, allowing easy navigation and management of charge points and RFID tokens.
- Flexible Database Support: Supports various database systems and provides scripts for easy setup.
- Enhanced Management Functions: Administrators have full control over the charge points and can generate and export detailed reports.

For more information and to access the source code, visit: [OCPP.Core on GitHub](https://github.com/dallmann-consulting/OCPP.Core)

https://github.com/dallmann-consulting/OCPP.Core

goekay, 2024-08-05:
since the user deleted his/her github account, all of his/her messages are gone. therefore, the comments here appear disconnected since the context is missing.

here is the PDF extract of the original exchange:
[steve-issue-1524-export.pdf](https://github.com/user-attachments/files/16495074/steve-issue-1524-export.pdf)

hasanr68, 2024-09-22:
@juherr i am thinking of implementing killbill, would you have used that or getlago? And watchouts or tips please?
Thanks,

juherr, 2024-09-22:
@hasanr68 I've never used them but they are in my todo list.
Please, give feedbacks if try them.

aramphibv, 2024-09-23:
Hi, I'm running the current github state and I have https working (cost me some fiddling with letsencrypt certs) but it's running.
I tried the swagger API interface but always get:
```
  "status": 401,
  "error": "Unauthorized",
  "message": "The user does not exist, exists but is disabled or has API access disabled.",
```

I click the auth button:
[image]
Used the regular admin username and pass and also tried with webapi.key as username and webapi.value as password

Any pointers here on how to get it working :-)?