steve-community/steve #1526: Validate chargeBoxId for WS connections

goekay, 2024-08-01:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : ...
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

Steve should not allow [any arbitrary last bit of URL](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/OcppWebSocketHandshakeHandler.java#L72) to enter the system. We actually have [ChargeBoxIdValidator](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/validation/ChargeBoxIdValidator.java) that can be used to make sure that this String is not in a problematic format.

#### Actual Behavior

...

#### Steps to Reproduce the Problem

  1. ...
  2. ...
  3. ...

#### Additional context

...

goekay, 2024-08-01:
The regex we are using allows any non-whitespace character which is too generous. I am thinking about applying some restrictions similar to [IdTagValidator](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/validation/IdTagValidator.java) but without the length constraint. 

wdyt @juherr

juherr, 2024-08-01:
@goekay I think it should be configurable and disabled by default. 
Otherwise it will break existing instances where blanks are currently used (by choice or error).

In case the option is not enabled, a warn log may be used for each stations registred with unexpected characters.

juherr, 2024-08-01:
@goekay why not having an option that sanitizes names and remove unexpected chars? It can help to follow recommandations without breaking things.

goekay, 2024-08-01:
> I think it should be configurable and disabled by default.
> Otherwise it will break existing instances where blanks are currently used (by choice or error).

i try to prevent configuration/property explosion. it would be great if a solution without property is possible.

> why not having an option that sanitizes names and remove unexpected chars? It can help to follow recommandations without breaking things.

because the "input" will differ from the "entry" in our system, and it might lead to confusion.

closed 2024-08-04 by goekay