steve-community/steve #1540: Store Web API key in database

goekay, 2024-08-12:
#### Checklist

- [x] I checked other issues already and found no answer
- [x] I checked the documentation and found no answer
- [x] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

currently, we are storing [web api key in properties file](https://github.com/steve-community/steve/blob/master/src/main/resources/config/prod/main.properties#L23-L24). this is problematic for multiple reasons: 
- it is not dynamic/flexible/easy-to-change since hardcoded as property
- it is a singleton value valid for all steve API access. it is the same for all types API users. a user _A_ and another user _B_ should not share the key

#### Describe the solution you'd like

since we are moving into database for multi-users and rbac (see #991, #1165 and #1539), i am thinking about adding another column `api_token` to `web_user`. this way, each user will have the possibility to access web UI and API.

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

goekay, 2024-08-17:
during the impl of https://github.com/steve-community/steve/pull/1545 i realized the following. 

- we have to switch from api key to basic auth: we were setting headers for api key like `STEVE-API-KEY: apiKey` so far. i can extract `apiKey` into database, but which user should i associate this with? so far, it was an application-global setting but not anymore. we could just search for `apiKey` within all users... and authenticate this call whenever we find a match. this would probably be a security issue. therefore, these keys need to be specific for users. which nicely fits the basic auth mechanism, and if we do the switch, the key is not a key anymore but rather a password.
- `api_token` in database actually represents an "api password". this is interesting, and allows an admin/manager to set different passwords for UI and API. both can be set to the same value. i see this as an added expressiveness. is this an overkill?

juherr, 2024-08-17:
It is logical to consider using Basic Authentication when treating the `api_token` as an API password.
However, as an API integrator, I have not encountered such a method of API authentication before. 

It would be beneficial to explore dedicated stacks specifically designed for this purpose, such as Keycloak and other similar options.

In my view, offering a default implementation that is quick and rudimentary can be helpful, as long as it doesn't limit the option of replacing it with a more secure and improved solution in the future.

goekay, 2024-08-18:
> However, as an API integrator, I have not encountered such a method of API authentication before.

really? i am integrating various APIs in my day job as well and i see all kinds of flavours: some fixed value in URL (no one should do that), custom header key/value, bearer token, basic auth, etc. the variations are so many, that i find the decision even _kind of insignificant_ (except the value/token in URL. no one should do that). basic auth will create a base64 token out of `username:pwd` anyways which will be sent over the wire.

switching between auth mechanisms in HTTP calls can this easy:

```
org.springframework.http.HttpHeaders headers = new HttpHeaders();
// opt 1: current impl in steve
headers.set("STEVE-API-KEY", "webapi key value"); 
// opt 2
headers.setBearerAuth("bearer token");
// opt 3
headers.setBasicAuth("username", "pwd");
```

my problem with custom header and bearer is that i cannot associate them with a `web_user`.

juherr, 2024-08-18:
I agree that it exist many ways to implement authentication in API, but there is a kind of consensus these days for oauth or JWT. 
I'm not a security expert, and that's why I'm interested in understanding what are the best practices and where they come from.
Do you have some public API examples that are using api_token as you proposed?

I don't get the security problem of having an access token in the webuser table. 
It can be seen as a denormalized way to avoid the token table.
Many examples on the web explain how to implement api_token. For example, https://medium.com/@sallu-salman/implementing-token-based-authentication-in-a-spring-boot-project-dba7811ffcee

If you want to keep it simple (and security is never simple), why not just remove the api_token concept and the user login/password with basicauth?
The current implementation is still valid as a simple service token.

goekay, 2024-08-18:
> Do you have some public API examples that are using api_token as you proposed?

if you google "Basic auth for REST APIs", you will see many _many_ results. here is one from atlassian: https://developer.atlassian.com/cloud/jira/platform/basic-auth-for-rest-apis/

in fact, they refer to their API pwd as "api token" as well:

> Generate an API token for Jira using your [Atlassian Account](https://id.atlassian.com/manage/api-tokens).

if you read between the lines, they distinguish between normal pwd and api pwd (i.e. api token), as well. this is also my proposition. 

> If you want to keep it simple (and security is never simple), why not just remove the api_token concept and the user login/password with basicauth?

see my comment from above:

> api_token in database actually represents an "api password". this is interesting, and allows an admin/manager to set different passwords for UI and API. both can be set to the same value. i see this as an added expressiveness. is this an overkill?

so, i actually thought about this, but then realized: if people want to use this way they can set both columns (password and api_token) to the same value. the solution i am providing can do _that_ but also more (different values).

juherr, 2024-08-18:
You got me, you found someone using the same solution.
That doesn't mean it is the recommended one or the current state of the art.

BTW, I think it is an overkill if you want something simple, and may not meet the desired level of perfection in terms of elegance.

goekay, 2024-08-18:
> That doesn't mean it is the recommended one or the current state of the art.

you wanted examples, i gave an example. but also mentioned there are many similar solutions/posts/results.

juherr, 2024-08-18:
You are correct. It seems that Atlassian is the only one that associates "login" and "token" in my first two pages of Google search results when I supposed there is none.

I think the use of the term "token" instead of "password" can be confusing. To avoid this confusion and any debate, a possible solution would be to rename it as "api_password".

goekay, 2024-08-18:
please do not get caught by the used terminology. sure, i can rename.

but would you agree with me that "basic auth for rest apis" is indeed common approach now? and that i am not doing something wild?

juherr, 2024-08-18:
I don't consider Basic as the best choice, but I have no strong argument against it.
It isn't worth debating which one is better because they are both good choices.

In my mind (and maybe only mine) token_api refers to Bearer or specific header.

If the variable is renamed password, there is no possible confusion and Basic will be the best choice in that case.

closed 2024-08-21 by goekay