steve-community/steve #1555: Which versions currently support pending status in boot notification responses?

warukunaiyone, 2024-08-22:
I want to test a respone of Boot Notification (OCPP 1.6J = Accepted, Pending, Rejected)

I saw from #202 already but I installed multiple versions in the create Charge Point Registration status page: it only has the option Accepted and Rejected.

So I would like to ask which versions currently support the pending status or how do I set it up before sending the pending status back to the charger?

[image]

juherr, 2024-08-22:
It should be 
[image]

goekay, 2024-08-22:
all three status values are supported. keep in mind the status "pending" came with 1.6. steve supports 1.5 and 1.2 as well.

if you add a new station, [this applies](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/controller/ChargePointsController.java#L231-L232). steve determines the ocpp protocol and the version with the first websocket connection.

afterwards, you can go to the details page of the station and toggle the status:

[image]

nikhilRailbit99, 2024-10-18:
How to connect and register steve implementation to my eureka micro service server ?