steve-community/steve #1569: Transaction Id is coming as 0 from charger on transaction stop and this creates an exception

sijojamesjohn, 2024-09-04:
#### Checklist

- [ ] I checked other issues already, but found no answer/solution
- [ ] I checked the documentation and wiki, but found no answer/solution
- [ ] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : Version:	3.6.0
Database Version:	1.0.3
Operating system  : Amazon Linux release 2 (Karoo)

JDK               : openjdk 11.0.23 2024-04-16 LTS

Database          : mysql
```

#### Expected Behavior
Find the last transaction id from meter values are use that?
...

#### Actual Behavior
[INFO ] 2024-09-04 12:13:15,829 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger (qtp432430300-19463) - [chargeBoxId=TZG0C460AB, sessionId=a006d37d-cf26-0e2a-ad00-3b6a02c055a2] Received: [2,"13845062","StopTransaction",{"idTag":"0000000000001500","meterStop":140857,"timestamp":"2024-09-03T18:52:30+01:00","transactionId":0,"reason":"Local","transactionData":[{"timestamp":"2024-09-03T18:52:30+01:00","sampledValue":[{"value":"140857","context":"Transaction.End","format":"Raw","measurand":"Energy.Active.Import.Register","unit":"Wh"}]}]}]
[ERROR] 2024-09-04 12:13:15,834 de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl (qtp432430300-19463) - Exception occurred
org.jooq.exception.DataAccessException: SQL [insert into `stevedb`.`transaction_stop` (`transaction_pk`, `event_timestamp`, `event_actor`, `stop_timestamp`, `stop_value`, `stop_reason`) values (?, ?, ?, ?, ?, ?)]; Cannot add or update a child row: a foreign key constraint fails (`stevedb`.`transaction_stop`, CONSTRAINT `FK_transaction_stop_transaction_pk` FOREIGN KEY (`transaction_pk`) REFERENCES `transaction_start` (`transaction_pk`) ON DELETE CASCADE ON UPDATE NO AC)
	at org.jooq_3.16.7.MYSQL.debug(Unknown Source) ~[?:?]
	at org.jooq.impl.Tools.translate(Tools.java:3095) ~[jooq-3.16.7.jar:?]
	at org.jooq.impl.DefaultExecuteContext.sqlException(DefaultExecuteContext.java:670) ~[jooq-3.16.7.jar:?]
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:354) ~[jooq-3.16.7.jar:?]
	at org.jooq.impl.AbstractDelegatingQuery.execute(AbstractDelegatingQuery.java:115) ~[jooq-3.16.7.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.updateTransaction(OcppServerRepositoryImpl.java:327) ~[steve.jar:?]
	at de.rwth.idsg.steve.service.CentralSystemService16_Service.stopTransaction(CentralSystemService16_Service.java:221) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.CentralSystemService16_SoapServer.stopTransaction(CentralSystemService16_SoapServer.java:117) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint$Ocpp16CallHandler.dispatch(Ocpp16WebSocketEndpoint.java:97) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.AbstractCallHandler.accept(AbstractCallHandler.java:46) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.IncomingPipeline.accept(IncomingPipeline.java:61) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleTextMessage(AbstractWebSocketEndpoint.java:120) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.onMessage(AbstractWebSocketEndpoint.java:91) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ConcurrentWebSocketHandler.handleMessage(ConcurrentWebSocketHandler.java:50) ~[steve.jar:?]
	at org.springframework.web.socket.adapter.jetty.Jetty10WebSocketHandlerAdapter.onWebSocketText(Jetty10WebSocketHandlerAdapter.java:78) ~[spring-websocket-5.3.23.jar:5.3.23]
	at org.eclipse.jetty.websocket.core.internal.messages.StringMessageSink.accept(StringMessageSink.java:53) ~[websocket-core-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.acceptMessage(JettyWebSocketFrameHandler.java:334) ~[websocket-jetty-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.onTextFrame(JettyWebSocketFrameHandler.java:422) ~[websocket-jetty-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.common.JettyWebSocketFrameHandler.onFrame(JettyWebSocketFrameHandler.java:240) ~[websocket-jetty-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession$IncomingAdaptor.lambda$onFrame$1(WebSocketCoreSession.java:671) ~[websocket-core-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.server.handler.ContextHandler.handle(ContextHandler.java:1470) ~[jetty-server-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.server.handler.ContextHandler.handle(ContextHandler.java:1489) ~[jetty-server-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.core.server.internal.AbstractHandshaker$1.handle(AbstractHandshaker.java:212) ~[websocket-core-server-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession$IncomingAdaptor.onFrame(WebSocketCoreSession.java:671) ~[websocket-core-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.core.internal.ExtensionStack.onFrame(ExtensionStack.java:120) ~[websocket-core-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.core.internal.WebSocketCoreSession.onFrame(WebSocketCoreSession.java:481) ~[websocket-core-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.onFrame(WebSocketConnection.java:262) ~[websocket-core-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.fillAndParse(WebSocketConnection.java:455) ~[websocket-core-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.websocket.core.internal.WebSocketConnection.onFillable(WebSocketConnection.java:340) ~[websocket-core-common-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:314) ~[jetty-io-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:100) ~[jetty-io-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.io.SelectableChannelEndPoint$1.run(SelectableChannelEndPoint.java:53) ~[jetty-io-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.runTask(AdaptiveExecutionStrategy.java:416) ~[jetty-util-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.consumeTask(AdaptiveExecutionStrategy.java:385) ~[jetty-util-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.tryProduce(AdaptiveExecutionStrategy.java:272) ~[jetty-util-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.util.thread.strategy.AdaptiveExecutionStrategy.lambda$new$0(AdaptiveExecutionStrategy.java:140) ~[jetty-util-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:411) ~[jetty-util-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:936) ~[jetty-util-10.0.14.jar:10.0.14]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$Runner.run(QueuedThreadPool.java:1080) ~[jetty-util-10.0.14.jar:10.0.14]
	at java.lang.Thread.run(Thread.java:829) [?:?]
Caused by: java.sql.SQLIntegrityConstraintViolationException: Cannot add or update a child row: a foreign key constraint fails (`stevedb`.`transaction_stop`, CONSTRAINT `FK_transaction_stop_transaction_pk` FOREIGN KEY (`transaction_pk`) REFERENCES `transaction_start` (`transaction_pk`) ON DELETE CASCADE ON UPDATE NO AC)
	at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:117) ~[mysql-connector-java-8.0.30.jar:8.0.30]
	at com.mysql.cj.jdbc.exceptions.SQLExceptionsMapping.translateException(SQLExceptionsMapping.java:122) ~[mysql-connector-java-8.0.30.jar:8.0.30]
	at com.mysql.cj.jdbc.ServerPreparedStatement.serverExecute(ServerPreparedStatement.java:555) ~[mysql-connector-java-8.0.30.jar:8.0.30]
	at com.mysql.cj.jdbc.ServerPreparedStatement.executeInternal(ServerPreparedStatement.java:339) ~[mysql-connector-java-8.0.30.jar:8.0.30]
	at com.mysql.cj.jdbc.ClientPreparedStatement.execute(ClientPreparedStatement.java:354) ~[mysql-connector-java-8.0.30.jar:8.0.30]
	at com.zaxxer.hikari.pool.ProxyPreparedStatement.execute(ProxyPreparedStatement.java:44) ~[HikariCP-5.0.1.jar:?]
	at com.zaxxer.hikari.pool.HikariProxyPreparedStatement.execute(HikariProxyPreparedStatement.java) ~[HikariCP-5.0.1.jar:?]
	at org.jooq.tools.jdbc.DefaultPreparedStatement.execute(DefaultPreparedStatement.java:219) ~[jooq-3.16.7.jar:?]
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:419) ~[jooq-3.16.7.jar:?]
	at org.jooq.impl.AbstractDMLQuery.execute(AbstractDMLQuery.java:961) ~[jooq-3.16.7.jar:?]
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:340) ~[jooq-3.16.7.jar:?]
	... 36 more
...

#### Steps to Reproduce the Problem

  1. Start transaction
  2. stop transaction without transaction id

#### Additional context

...

DeadBoySmile, 2024-09-04:
try this
https://gofile.io/d/Lji8KD
select the desired compiler in the installer

juherr, 2024-09-04:
@sijojamesjohn 
What is your charger and firmware version?
Is it referenced in the compatibility list? https://github.com/steve-community/steve/wiki/Charging-Station-Compatibility

BTW, that is not a Steve issue because the charger is not supposed to provide 0 as transactionId:
[image]

You'll need to contact the manufacturer and ask them for a fix.

sijojamesjohn, 2024-09-09:
OK thanks for the info. Can go back to the client with this.

closed 2024-09-09 by sijojamesjohn