steve-community/steve #1627: Adding calculated Consumption for Transactions Table

MXIII, 2024-11-06:
#### Checklist

- [X ] I checked other issues already and found no answer
- [X ] I checked the documentation and found no answer
- [X ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve
In the Transactions-Overview we have a starting and an end value for charging.
Its a little bit of annoying to count the consumption manually but an easy fix would be to calculate the consumption next to the start-value and end-value just by substracting the startvalue from the endvalue.

the transactionspage needs an overhaul and restructuring, because the start and enddate should be next to each other and the startvalue and endvalue + consumption too, for better comparison, we might also would want to add the username instead of just the ocppTagId to the table.

`            <tr>
                <th data-sort="int">Transaction ID</th>
                <th data-sort="string">Wallbox ID</th>
                <th data-sort="int">Connector ID</th>
                <th data-sort="string">Tag ID</th>
	         <th data-sort="string">Username</th>			
                <th data-sort="date">Started</th>
                <th data-sort="date">Ended</th>
                <th data-sort="int">Start Value</th>
                <th data-sort="int">End Value</th>
                <th data-sort="string">Consumption (W)</th>
		<th data-sort="string">Stop Reason</th>
                <th></th>
            </tr>
`
...

#### Describe the solution you'd like
In [image]y transactions.jsp i have added the following line:
<td>${ta.stopValue - ta.startValue}</td>
and added a table header with "consumption (w)"
it might not be reliable (i dont know yet) but it works for me currently and i have a total consumption in watts in the transactionstable
maybe there is a better way to solve this issue.

...

...

#### Additional context
i would also like to add the username to the table next to the ocppTagId instead of just the ocppTagId.
added a screenshot how it looks in my deployment
...

goekay, 2024-11-17:
i could swear there was a work-in-progress PR by a contributor who was trying to do this, but i could not find it right now.

basically, he did some initial work but there were some edge cases which i pointed out, but he did not continue. so PR died i guess.

nikhilkandari77, 2025-08-30:
How I Handle Missed Stop Events ->

If, for some reason, a stop event is not received from the charge point, the only fallback is to use the latest MeterValues packet. To handle this in a robust way:

I maintain a transaction ID queue where every incoming meter value packet updates the corresponding transaction.

A background cron job periodically checks this queue to determine whether transactions have become stale (i.e., no new meter values for a defined timeout).

Such transactions are marked as "stop event not received" in reports, ensuring transparency that they were closed by inference rather than by a proper OCPP stop event.

This way, the backend can gracefully handle abnormal sessions while still being compliant with OCPP semantics.

Dual-Gun Charging Specifics ->

Dual-gun chargers add complexity:

Some implementations use one transaction ID across both connectors (shared session).

Others use independent transaction IDs per connector (the clean OCPP-compliant approach).

If there is one shared transaction ID:

The backend cannot close the session until both connectors stop sending meter values.

Otherwise, one active gun could be prematurely cut off.

Therefore, the only safe option is again to rely on the latest meter value packets, combined with inactivity timeouts, before marking the transaction as closed.

juherr, 2025-09-01:
@goekay Oops, sorry, I hadn’t noticed that before. It looks like Steve already has a solid UI. 😉