steve-community/steve #1638: Help for Steve

andy-prn, 2024-11-21:
Hello,

I have an issue with setting up Steve. I created a VM on Proxmox with Ubuntu Server installed. I successfully installed Steve, and I can access the web interface and obtain the WebSocket.

On Steve's web interface, I create my charging station with an ID like 'B1', for example.

Then I go to my charging station, connect to its interface via Wi-Fi, and enter:

    - ID: B1
    - WebSocket: ws://192.168.49.116:8086/steve/websocket/CentralSystemService
    - Port: 8086

But here’s where the problem occurs: no data is coming through to Steve.
However, I can ping my charging stations from my server.

Could you please help me?
I am French, so I translated this text; I apologize in advance for any mistakes.

juherr, 2024-11-21:
Salut,

Le bugtracker n'est pas dédié au support technique mais aux bugs.
Néanmoins, tu devrais essayer de limiter la complexité avant l'épreuve du feu : 
* vérifier que la connexion fonctionne via Postman ou un simulateur de borne, ou
* vérifier que la borne se connecte bien sur une instance de Steve local à ton pc

Si possible, peux-tu communiquer la marque, modèle et version firmware de ta borne ?
Enfin, si tu peux partager un screenshot de l'interface de configuration de la borne, cela aiderait à t'aider.

andy-prn, 2024-11-22:
Hello, 

Désolé mais je ne savais pas où/comment écrire pour avoir un peu d'aide.

Alors désolé mais je ne connais pas du Postman ... je ne sais pas trop comment ça marche.
Les bornes sont des Metron Double [https://wattislove.com/fr/recharge-ve/borne-de-recharge-double-2x7-4kw-2x22kw/?srsltid=AfmBOor4uV9jsXjfz6QMhN-dtIq1c_ArXW7LBrHxLMQbljz_7xQ0Vodi](url)
Elles sont en OCPP V1.5
Et voici les screenshots de l'interface de la borne.
Merci de ton aide.
[image]
[image]

juherr, 2024-11-22:
@andy-prn 
A la vue de l'interface, je pense que la borne construit elle-même l'url et qu'il n'est peut-être pas possible de la faire fonctionner avec Steve.

La borne doit se connecter avec l'url suivante :
ws://192.168.49.116:8086/steve/websocket/CentralSystemService/B1

Donc
`ws://192.168.49.116` => hostname
`8086` => port
`B1` => UUID
mais il reste à trouver où mettre `steve/websocket/CentralSystemService`.

A titre personnel, je testerai 
* soit : 
`ws://192.168.49.116` => hostname
`8086/steve/websocket/CentralSystemService` => port
`B1` => UUID
* soit : 
`ws://192.168.49.116` => hostname
`8086` => port
`steve/websocket/CentralSystemService/B1` => UUID

Il faut contacter le support du fabricant car ils sont les mieux placés pour répondre à ce genre de questions.

Aussi, il est préférable de fermer le ticket car il s'agit ici d'un soucis de la borne et pas de Steve.

andy-prn, 2024-11-22:
Hello,

Merci pour la réponse je viens de tester et ta solution 2 fonctionne !!! :) grand merci encore.

Je clos le ticket.

closed 2024-11-22 by andy-prn