steve-community/steve #1644: Error when compiling with OpenJDK 23

Smoerrebroed, 2024-11-28:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.7.1 but also happens with current Git version
Operating system  : Arch Linux
JDK               : OpenJDK 23
Database          : MariaDB 11.x (not part of the issue)
```

#### Expected Behavior

./mvnw package produces an executable version of SteVe

#### Actual Behavior

It throws lots of errors all pointing to some missing dependency/declaration like the following:

[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /home/xxx/Projekte/steve/src/main/java/de/rwth/idsg/steve/Application.java:[41,9] Symbol nicht gefunden
  Symbol: Variable log
  Ort: Klasse de.rwth.idsg.steve.Application
[ERROR] /home/xxx/Projekte/steve/src/main/java/de/rwth/idsg/steve/Application.java:[41,77] Symbol nicht gefunden
  Symbol: Methode getProfile()
  Ort: Variable sc von Typ de.rwth.idsg.steve.SteveConfiguration
...

#### Steps to Reproduce the Problem

  1. Install/activate OpenJDK 23
  2. Clone SteVe
  3. ./mvnw package

#### Additional context

I switched to OpenJDK 21 and that works like it used to.

goekay, 2024-11-29:
hey, thanks for this report. it is indeed a problem. thankfully, the fix was simple. please see the respective PR, test and give feedback that it resolves your issue.

closed 2024-12-01 by goekay

Smoerrebroed, 2024-12-03:
Hi @goekay , apologies for the delay. I just got around to testing the fix and while it fixes the compile error, SteVe no longer starts up afterwards. This is independent of using Java 21 or Java 23 (or any permutation of using these for compilation/running). The error message I get is:

`[ERROR] 2024-12-03 11:54:08,304 org.springframework.web.context.ContextLoader (main) - Context initialization failed
org.springframework.beans.factory.BeanDefinitionStoreException: Failed to read candidate component class: URL [jar:file:/home/xxx/steve/target/libs/springdoc-openapi-starter-common-2.7.0.jar!/org/springdoc/core/configuration/SpringDocSpecPropertiesConfiguration.class]
	at org.springframework.context.annotation.ClassPathScanningCandidateComponentProvider.scanCandidateComponents(ClassPathScanningCandidateComponentProvider.java:510) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ClassPathScanningCandidateComponentProvider.findCandidateComponents(ClassPathScanningCandidateComponentProvider.java:351) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ClassPathBeanDefinitionScanner.doScan(ClassPathBeanDefinitionScanner.java:277) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ComponentScanAnnotationParser.parse(ComponentScanAnnotationParser.java:128) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ConfigurationClassParser.doProcessConfigurationClass(ConfigurationClassParser.java:306) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ConfigurationClassParser.processConfigurationClass(ConfigurationClassParser.java:246) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ConfigurationClassParser.parse(ConfigurationClassParser.java:197) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ConfigurationClassParser.parse(ConfigurationClassParser.java:165) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ConfigurationClassPostProcessor.processConfigBeanDefinitions(ConfigurationClassPostProcessor.java:417) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ConfigurationClassPostProcessor.postProcessBeanDefinitionRegistry(ConfigurationClassPostProcessor.java:290) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.support.PostProcessorRegistrationDelegate.invokeBeanDefinitionRegistryPostProcessors(PostProcessorRegistrationDelegate.java:349) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.support.PostProcessorRegistrationDelegate.invokeBeanFactoryPostProcessors(PostProcessorRegistrationDelegate.java:118) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.support.AbstractApplicationContext.invokeBeanFactoryPostProcessors(AbstractApplicationContext.java:789) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:607) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.web.context.ContextLoader.configureAndRefreshWebApplicationContext(ContextLoader.java:394) ~[spring-web-6.1.11.jar:6.1.11]
	at org.springframework.web.context.ContextLoader.initWebApplicationContext(ContextLoader.java:274) ~[spring-web-6.1.11.jar:6.1.11]
	at org.springframework.web.context.ContextLoaderListener.contextInitialized(ContextLoaderListener.java:102) ~[spring-web-6.1.11.jar:6.1.11]
	at org.eclipse.jetty.ee10.servlet.ServletContextHandler.callContextInitialized(ServletContextHandler.java:1587) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.ee10.servlet.ServletContextHandler.contextInitialized(ServletContextHandler.java:498) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.ee10.servlet.ServletHandler.initialize(ServletHandler.java:670) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.ee10.servlet.ServletContextHandler.startContext(ServletContextHandler.java:1321) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.ee10.webapp.WebAppContext.startWebapp(WebAppContext.java:1346) ~[jetty-ee10-webapp-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.ee10.webapp.WebAppContext.startContext(WebAppContext.java:1304) ~[jetty-ee10-webapp-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.ee10.servlet.ServletContextHandler.lambda$doStart$0(ServletContextHandler.java:1047) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.server.handler.ContextHandler$ScopedContext.call(ContextHandler.java:1452) ~[jetty-server-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.ee10.servlet.ServletContextHandler.doStart(ServletContextHandler.java:1044) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.ee10.webapp.WebAppContext.doStart(WebAppContext.java:503) ~[jetty-ee10-webapp-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:93) ~[jetty-util-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:169) ~[jetty-util-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:120) ~[jetty-util-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.server.Handler$Abstract.doStart(Handler.java:491) ~[jetty-server-12.0.15.jar:12.0.15]
	at org.eclipse.jetty.server.handler.gzip.GzipHandler.doStart(GzipHandler.java:132) ~[jetty-server-12.0.15.jar:12.0.15]
[... cut ...]
	at org.springframework.context.annotation.ClassPathScanningCandidateComponentProvider.isCandidateComponent(ClassPathScanningCandidateComponentProvider.java:547) ~[spring-context-6.1.11.jar:6.1.11]
	at org.springframework.context.annotation.ClassPathScanningCandidateComponentProvider.scanCandidateComponents(ClassPathScanningCandidateComponentProvider.java:471) ~[spring-context-6.1.11.jar:6.1.11]
	... 53 more
`
Using the same steps with the latest release (on Java 21), there is no error message. Any ideas?

goekay, 2024-12-03:
thx for the report. i found the problem. it was not about jdk 23. will create a new issue for better tracking, since it is not related.

commit 6467442f (2024-11-29): add annotationProcessorPaths (closes #1644)

builds were working without this until jdk 23 (for whatever reason).
it is part of lombok documentation to use this anyways: https://projectlombok.org/setup/maven
so we are adding it.
commit 28538f11 (2024-12-01): Merge pull request #1645 from steve-community/1644-error-when-compiling-with-openjdk-23

add annotationProcessorPaths (closes #1644)