steve-community/steve #1648: steve fails to start after openapi-starter-webmvc-ui update

goekay, 2024-12-03:
#### Checklist

- [x] I checked other issues already, but found no answer/solution
- [x] I checked the documentation and wiki, but found no answer/solution
- [x] I am running the latest version and the issue still occurs
- [x] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : HEAD of the repo
Operating system  : ...
JDK               : ...
Database          : ...
```

#### Expected Behavior

steve starts and runs successfully.

#### Actual Behavior

the application crashes with the following

```
[ERROR] 2024-12-03 11:54:08,304 org.springframework.web.context.ContextLoader (main) - Context initialization failed org.springframework.beans.factory.BeanDefinitionStoreException: 
Failed to read candidate component class: URL [jar:file:/home/xxx/steve/target/libs/springdoc-openapi-starter-common-2.7.0.jar!/org/springdoc/core/configuration/SpringDocSpecPropertiesConfiguration.class] 
at org.springframework.context.annotation.ClassPathScanningCandidateComponentProvider.scanCandidateComponents(ClassPathScanningCandidateComponentProvider.java:510) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ClassPathScanningCandidateComponentProvider.findCandidateComponents(ClassPathScanningCandidateComponentProvider.java:351) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ClassPathBeanDefinitionScanner.doScan(ClassPathBeanDefinitionScanner.java:277) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ComponentScanAnnotationParser.parse(ComponentScanAnnotationParser.java:128) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ConfigurationClassParser.doProcessConfigurationClass(ConfigurationClassParser.java:306) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ConfigurationClassParser.processConfigurationClass(ConfigurationClassParser.java:246) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ConfigurationClassParser.parse(ConfigurationClassParser.java:197) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ConfigurationClassParser.parse(ConfigurationClassParser.java:165) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ConfigurationClassPostProcessor.processConfigBeanDefinitions(ConfigurationClassPostProcessor.java:417) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ConfigurationClassPostProcessor.postProcessBeanDefinitionRegistry(ConfigurationClassPostProcessor.java:290) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.support.PostProcessorRegistrationDelegate.invokeBeanDefinitionRegistryPostProcessors(PostProcessorRegistrationDelegate.java:349) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.support.PostProcessorRegistrationDelegate.invokeBeanFactoryPostProcessors(PostProcessorRegistrationDelegate.java:118) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.support.AbstractApplicationContext.invokeBeanFactoryPostProcessors(AbstractApplicationContext.java:789) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:607) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.web.context.ContextLoader.configureAndRefreshWebApplicationContext(ContextLoader.java:394) ~[spring-web-6.1.11.jar:6.1.11] 
at org.springframework.web.context.ContextLoader.initWebApplicationContext(ContextLoader.java:274) ~[spring-web-6.1.11.jar:6.1.11] 
at org.springframework.web.context.ContextLoaderListener.contextInitialized(ContextLoaderListener.java:102) ~[spring-web-6.1.11.jar:6.1.11] 
at org.eclipse.jetty.ee10.servlet.ServletContextHandler.callContextInitialized(ServletContextHandler.java:1587) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.ee10.servlet.ServletContextHandler.contextInitialized(ServletContextHandler.java:498) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.ee10.servlet.ServletHandler.initialize(ServletHandler.java:670) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.ee10.servlet.ServletContextHandler.startContext(ServletContextHandler.java:1321) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.ee10.webapp.WebAppContext.startWebapp(WebAppContext.java:1346) ~[jetty-ee10-webapp-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.ee10.webapp.WebAppContext.startContext(WebAppContext.java:1304) ~[jetty-ee10-webapp-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.ee10.servlet.ServletContextHandler.lambda$doStart$0(ServletContextHandler.java:1047) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.server.handler.ContextHandler$ScopedContext.call(ContextHandler.java:1452) ~[jetty-server-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.ee10.servlet.ServletContextHandler.doStart(ServletContextHandler.java:1044) ~[jetty-ee10-servlet-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.ee10.webapp.WebAppContext.doStart(WebAppContext.java:503) ~[jetty-ee10-webapp-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:93) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:169) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:120) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.server.Handler$Abstract.doStart(Handler.java:491) ~[jetty-server-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.server.handler.gzip.GzipHandler.doStart(GzipHandler.java:132) ~[jetty-server-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:93) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:169) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:120) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.server.Handler$Abstract.doStart(Handler.java:491) ~[jetty-server-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.server.handler.ContextHandler.lambda$doStart$0(ContextHandler.java:754) ~[jetty-server-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.server.handler.ContextHandler$ScopedContext.call(ContextHandler.java:1452) ~[jetty-server-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.server.handler.ContextHandler.doStart(ContextHandler.java:754) ~[jetty-server-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:93) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:169) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:120) ~[jetty-util-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.server.Handler$Abstract.doStart(Handler.java:491) ~[jetty-server-12.0.15.jar:12.0.15] 
at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:93) ~[jetty-util-12.0.15.jar:12.0.15] 
[... 5 lines cut ...]
at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:93) ~[jetty-util-12.0.15.jar:12.0.15] 
at de.rwth.idsg.steve.JettyServer.start(JettyServer.java:168) ~[steve.jar:?] 
at de.rwth.idsg.steve.SteveProdStarter.start(SteveProdStarter.java:55) ~[steve.jar:?] 
at de.rwth.idsg.steve.Application.start(Application.java:76) ~[steve.jar:?] 
at de.rwth.idsg.steve.Application.main(Application.java:64) ~[steve.jar:?] 

Caused by: java.lang.NoSuchFieldError: Class org.springframework.util.SystemPropertyUtils does not have member field 'java.lang.Character ESCAPE_CHARACTER' 
at org.springframework.boot.context.properties.bind.PropertySourcesPlaceholdersResolver.<init>(PropertySourcesPlaceholdersResolver.java:51) ~[spring-boot-3.4.0.jar:3.4.0] 
at org.springframework.boot.context.properties.bind.PropertySourcesPlaceholdersResolver.<init>(PropertySourcesPlaceholdersResolver.java:41) ~[spring-boot-3.4.0.jar:3.4.0] 
at org.springframework.boot.context.properties.bind.Binder.get(Binder.java:537) ~[spring-boot-3.4.0.jar:3.4.0] at org.springframework.boot.context.properties.bind.Binder.get(Binder.java:523) ~[spring-boot-3.4.0.jar:3.4.0] 
[... cut ...]
at org.springframework.context.annotation.ClassPathScanningCandidateComponentProvider.isCandidateComponent(ClassPathScanningCandidateComponentProvider.java:547) ~[spring-context-6.1.11.jar:6.1.11] 
at org.springframework.context.annotation.ClassPathScanningCandidateComponentProvider.scanCandidateComponents(ClassPathScanningCandidateComponentProvider.java:471) ~[spring-context-6.1.11.jar:6.1.11] 
... 53 more
```

#### Steps to Reproduce the Problem

  1. [pull the current HEAD of the repo](https://github.com/steve-community/steve/commit/8cdb790ef64171df03e030dd269a0ec0c813c71a)
  2. start the app
  3. it will fail

goekay, 2024-12-03:
the reason is the update in https://github.com/steve-community/steve/pull/1646. springdoc-openapi-starter-webmvc-ui moved to spring boot 3.4.0. [spring boot 3.4.0 depends on spring 6.2](https://github.com/spring-projects/spring-boot/wiki/Spring-Boot-3.4-Release-Notes#dependency-upgrades) but we still use 6.1.11.

the updated `PropertySourcesPlaceholdersResolver` started referencing some field `ESCAPE_CHARACTER` from `org.springframework.util.SystemPropertyUtils` which does not exist in spring 6.1.11.

[image]

updating the spring version resolves the issue.

moreover, our builds did not detect this, because open-api docs [are only activated on prod profile](https://github.com/steve-community/steve/blob/8cdb790ef64171df03e030dd269a0ec0c813c71a/src/main/java/de/rwth/idsg/steve/config/ApiDocsConfiguration.java#L55). i will remove this nuance, such that they are always there.

juherr, 2024-12-04:
@goekay maybe the ci should test all profiles on the recommended platform, right?

Smoerrebroed, 2024-12-04:
Thanks @goekay for the quick action. Looking forward to the next update. :-)

goekay, 2024-12-04:
@juherr the value added would be small.

we already have [a build matrix](https://github.com/steve-community/steve/blob/master/.github/workflows/main.yml#L9-L11) with 20 jobs. adding prod or not (another dimension with 2 values) would bring this job size to 40.

i instead removed this condition in my PR.

juherr, 2024-12-04:
I agree the value-add is small, but each profile is a distinct release target too.
That's why I proposed adding a new matrix with the recommended infrastructure and all profiles instead of adding a new variation in the existing matrix.
It should only add one job by profile under a non-regression test.

goekay, 2024-12-04:
@juherr sure, theoretically, i have nothing against this. however, it then means that we have to make decisions and pick [one os, one java version and one database](https://github.com/steve-community/steve/blob/master/.github/workflows/main.yml#L9-L11). what if there is an issue with the "other os/java/database and profile" combination? this approach would miss it. that being said, if you want to work on this, a PR would be welcome. technically, [we have this many profiles](https://github.com/steve-community/steve/tree/master/src/main/resources/config).

previously, i was including [security configs](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/config/SecurityConfiguration.java) and [openapi docs](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/config/ApiDocsConfiguration.java) only in PROD profile just for ease of development and DX: dont want to login each time when i am in fast iteration with frontend changes. dont want to add openapi code inspection which delays startup time by some seconds. having them only in PROD meant that our automated builds (which were using [TEST profile](https://github.com/steve-community/steve/blob/master/.github/workflows/main.yml#L40)) missed them. maybe another solution could be this: having these two aspects (security, openapi) enabled for PROD and TEST both. that way the DEV profile can still benefit from ease of development. WDYT?

closed 2024-12-07 by goekay

Smoerrebroed, 2024-12-07:
Works like a charm with Java 23 now. Again: Many thanks for the quick turnaround!