steve-community/steve #1650: Charging profiles to be updated at any time

huntc, 2024-12-06:
It appears that Steve doesn't permit a charging profile to be updated if there's a charging station associated with it. My understanding of OCPP is that charging profiles can change at any time given load management. Thoughts?

goekay, 2025-01-05:
my memory is a bit fuzzy in this subdomain, since it's been a while since i worked on it and looked at the spec. however, i remember the following thinking of mine:

i made a charging profile _instance_ (let's say the profile with id `24`) immutable, because otherwise (if it were mutable) it would be really hard to keep track of what configuration a profile with id `24` had at a given time, if users start to change this particular profile instance as they desire over time and across multiple stations. i wanted to prevent a chaotic scenario of not knowing the profile configuration with id `24` deterministically. 

> My understanding of OCPP is that charging profiles can change at any time given load management. 

charging behaviour can change with load management, sure. but this is not necessarily tied to having _mutable_ charging profile instances. ocpp allows adding, deleting, stacking of multiple profiles. these operations can be used to create complex scenarios of load management. the charging station would overlay them and create the composite charging profile which is the final and effective charging behaviour.

if you want to modify a particular charging profile instance with id `24`, the way to do it with the current implementation would be the following:
- clear charging profile `24` at stations [^1]
- create the new charging profile with the desired new configuration and id `25`
- set the charging profile `25` at stations

[^1]: this step is not even needed, if you use the same stack level in the new charging profile instance. spec says: _"Whenever a Charge Point receives a Charging Profile with a stackLevel and Purpose that already exists in the Charge Point, the Charge Point SHALL replace the existing profile."_

goekay, 2025-02-16:
hey @huntc is your question answered?