steve-community/steve #1664: Jetty is not taking the config main.properties

Yoken0S, 2024-12-23:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [X] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.7.1
Operating system  : Red8
JDK               : 21
Database          : mysql 8
```

Despite I'm changing the main.properties of prod as I'm not using a docker the server is running but keep running on IP 127.0.0.1 and port 8080 despite I'm changing the host.server and host.port

# Jetty configuration
#
server.host = 0.0.0.0
server.gzip.enabled = true

# Jetty HTTP configuration
#
http.enabled = true
http.port = 8080

# Jetty HTTPS configuration
#
https.enabled = false
https.port = 8443
keystore.path =
keystore.password =

Am I missing something ?

(I've disabled firewalld to test, same issue)

juherr, 2024-12-24:
Why do you say it is not working?
How can we reproduce the issue?

`server.host = 0.0.0.0` means "listen on all interfaces".

You can test on your own computer by calling the private adress instead of the local one.

Yoken0S, 2024-12-24:
It is not working because when I try to access through local (LAN) IP I've either a refused connection or a timeout connection.
There is no firewall on my LAN network.

And what I do not understand is when I start the server it says it is reachable from http://127.0.0.1:8080 even when I change IP and/or port.

goekay, 2024-12-24:
you need to rebuild/recompile the java application for your changes to have effect. are you doing that?

Yoken0S, 2024-12-24:
Omg yes sorry, so long I didn't play with that.

Thanks

closed 2024-12-24 by Yoken0S