steve-community/steve #1665: Issue with OCPP selection option when adding a charge point - Unable to choose OCPP type

FormelecIRVE, 2024-12-31:
Hi

I am encountering an issue when adding charge points via the SteVe web interface. When adding a charge point, I'm unable to select the OCPP type (e.g., OCPP 1.5, 1.6, etc.) in the add form. This field appears inactive, and I do not have the option to choose the desired communication mode.

I have checked the configuration and tried restarting the application, but the issue persists. The rest of the form seems functional, but this OCPP selection parameter is crucial for configuring the charge points.

Has anyone encountered this issue or could suggest any leads to solve it?

Thanks in advance for your help.

JJ
[image]

juherr, 2024-12-31:
Hi @FormelecIRVE 

You don't choose these values, it comes from the charger when it boots.

FormelecIRVE, 2024-12-31:
Thank you so much for responding 👍 

OK, it makes sens. I gonna try to manage with a virtual charger.

seehma, 2025-01-04:
iam also new here, but i got it to work on docker
1.) First enter your new wallbox in your Steve manager and choose a wallbox id (e.g. asdf)
2.) Enter the backend url of steve in your wallbox (in mine its called WebSockets JSON OCPP URL of Backend) should start with ws://192.168.1.250:8180/steve/websocket/CentralSystemService/asdf, also enter the same wallbox id and reboot the wallbox
3.) after some seconds it should show all information of the wallbox in your charging-point-list including ocpp version and other metadata...

goekay, 2025-01-11:
i consider this isse being resolved, and am closing it. you can re-open if you feel otherwise.

closed 2025-01-11 by goekay