steve-community/steve #1693: Active transaction

rdc-Green, 2025-02-13:
I have a charge point that has been offline for a number of months. The charge point was still being used but not being recorded in the database. Now that the charge point is back online i'm getting transactions come through, which i am stopping using the stop button, however the 'Active Transaction' count on the main page is still counting them.

Could you confirm where this data is pulled from in the database so that i can manually adjust it to remove these old non active transactions.

Thank you

juherr, 2025-02-13:
@rdc-Green Juste remove lines from `transaction_*` and `connector_meter_value` tables, and it should work as expected.

rdc-Green, 2025-02-14:
Thank you @juherr

closed 2025-02-14 by rdc-Green