steve-community/steve #1731: Not Showing Intermediate Meter Values

Dimitris1090T, 2025-03-25:
#### Checklist

- [X] I checked other issues already, but found no answer/solution
- [X] I checked the documentation and wiki, but found no answer/solution
- [X] I am running the latest version and the issue still occurs
- [ ] I am sure that this issue is about SteVe (and not about the charging station software or something unrelated to SteVe)

#### Specifications
```
SteVe Version     : 3.7.1
Operating system  : Ubuntu 22.04.5 LTS
JDK               : openjdk 17.0.14 2025-01-21
Database          : mariadb  Ver 15.1 Distrib 10.6.18-MariaDB
```

#### Problem
I connect Steve CSMS with SAP Simulation. In SAP install two templates of charging station, Kebba and ABB.
In Transaction View like you see bellow the intermediate Meter Values not display, but in Database i see normal all the meter values of this Transaction.
Why this happened, is bug in this version of STEVE or the SAP not send the correct data.

[image]

[image]

goekay, 2025-03-26:
this data behaviour seems to be conflicting with out "is this energy value" logic in [here](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java#L39-L85). and because of that, all meter values are filtered out.

the rows with "Wh" unit contain no other columns. this is almost in line with [this particular step](https://github.com/steve-community/steve/blob/master/src/main/java/de/rwth/idsg/steve/utils/TransactionStopServiceHelper.java#L54-L64), but we expect all columns to be null instead of all but one.

Dimitris1090T, 2025-03-26:
Thank you very much, i appreciate your help, that was the problem.
I changed logical operator from AND to OR and the meter values appear.

[image]

closed 2025-03-26 by Dimitris1090T

goekay, 2025-03-26:
we should fix this in steve though, if this is really a problem.

what SAP simulation are you using? is this behaviour of the stations OCPP-conform?

Dimitris1090T, 2025-03-26:
I'm using the SAP/e-mobility-charging-stations-simulator and the station templates I tested for connecting to Steve are the Keba (AC) station and the ABB (DC) station with OCPP 1.6 protocol. They connected without a problem and send meter values every 30 seconds after i click Start Transaction.

goekay, 2025-03-26:
could you pls create a bug ticket at https://github.com/SAP/e-mobility-charging-stations-simulator because, as far as i can see, steve follows the OCPP spec with regards to this, and they did not.

Dimitris1090T, 2025-03-26:
Thank you very much for your time and your advice, i create a bug ticket at https://github.com/SAP/e-mobility-charging-stations-simulator