steve-community/steve #1733: Migrate to Spring Boot

goekay, 2025-03-28:


juherr, 2025-03-28:
What about the data layer? 
Move to Spring data or keep jooq?

goekay, 2025-03-29:
i don't aim to rewrite controller, service or repository layers. the goal is to use spring boot's easy bootstrapping instead of hand-wiring. plus, jooq has enough/good spring boot interop.

juherr, 2025-03-29:
@andortoth As you already did the job once, do you have any advice or feedback that can help?

juherr, 2025-03-29:
@goekay Could you review [this commit](https://github.com/parklapp/steve-pluggable/commit/4265bdee62569cc1eb67cab238043fa4b4aad8ac) and let me know what should be backported?
From what I understand, the main challenge in migrating to Spring Boot is the startup process involving configuration.

goekay, 2025-03-29:
this is such a heavy commit. need to do it later (since it needs attention/time).

> main challenge in migrating to Spring Boot is the startup process involving configuration.

yeah, but most of the config is coming from spring beans and `@Configuration`s anyways. this does not change when we have spring boot. the main challenge should be the following.

currently, whenever steve starts (Application's main method), it starts a plain Jetty server(!). within that we start and configure spring (in SteveAppContext). so, spring's context is only living in a small world.

spring likes to be the mothership though. it should be the first thing to start, and everything else should live within that. so we have to change
- Application -> Jetty -> Spring

to
- Application -> Spring -> Jetty (implicitly, by spring boot)

---

as a scratchpad, i managed to migrate steve to spring boot in a really dirty testing-the-waters way, and then threw it away. spring boot owned everything, jetty was there, DB via jooq as well. app was starting successfully.

so, as i said, as long as we don't touch the layers and change business logic (this heavy commit also does other things), this should not be that hard.

goekay, 2025-03-29:
to reduce complexity and keep the scope focused, i dont aim to touch the enum class `SteveConfiguration` and change it to a spring bean. not yet. not within this one.

andortoth, 2025-03-29:
> [@andortoth](https://github.com/andortoth) As you already did the job once, do you have any advice or feedback that can help?

our version was a major overhaul, because of some business requirements that might not suit the Steve library and its design philosophy, so we decided not to make a pull request of it

the biggest differences are that we needed to replace JOGL with JPA because we needed to use the library with Azure SQL, and we had to be able to support a multi instance environment in a Kubernetes cluster so I don't really recommend to use our solution unless you have the same requirements (that is database platform independence or multi-instance support), because merging Steve fixes is quite a bit of fun with this amount of changes

if I remember correctly migrating to Spring Boot was not such a big deal, basically it was adding a @SpringBootApplication class, so if it's your only goal, I suggest forking the original library and configuring it to use Spring Boot, because our version as @goekay said, does a lot of other stuff as well

juherr, 2025-03-30:
@goekay 

> ⚠️ “If possible, JSPs should be avoided. There are several [known limitations](https://docs.spring.io/spring-boot/reference/web/servlet.html#web.servlet.embedded-container.jsp-limitations) when using them with embedded servlet containers.”

📄 Source: [Spring Boot Reference Documentation – JSP limitations](https://docs.spring.io/spring-boot/docs/current/reference/html/web.html#web.servlet.spring-mvc.template-engines)

Some other moves to consider:
* Replace Jetty with Tomcat (Spring Boot’s default embedded container)
* Replace JSP with Thymeleaf (Spring Boot’s default template engine)

goekay, 2025-09-22:
> ... i dont aim to touch the enum class SteveConfiguration and change it to a spring bean. not yet. not within this one.

contrary to what i said earlier, i did it within the corresponding PR and with commit https://github.com/steve-community/steve/pull/1822/commits/ddf7e3a178c95b98924e940a8f6ca6fa02aba400. after moving some properties into spring boot native properties, whatever was left was so little such that the effort was little.

closed 2025-09-24 by goekay