steve-community/steve #1750: Dose Steve support OCPP2.0.1/OCPP2.1 now?

WNmrwu, 2025-04-17:
#### Checklist

- [ ] I checked other issues already and found no answer
- [ ] I checked the documentation and found no answer
- [ ] I am running the latest version and the feature i am requesting is not implemented there

#### Describe the problem you are trying to solve

...

#### Describe the solution you'd like

...

#### Describe alternative solutions or features you've considered 

...

#### Additional context

...

juherr, 2025-04-17:
No

MeKeCNC, 2025-06-08:
Are there any plans for Steve to support OCPP2.0.1/OCPP2.1?

goekay, 2025-06-25:
> Are there any plans for Steve to support OCPP2.0.1/OCPP2.1?

yes, there are. there is a lot of thinking and planning happening behind the scenes right now, especially after the [Powerfill announcement](https://github.com/steve-community/steve/issues/1643).

prfcosta, 2025-09-15:
Hi,

what a date for plan lanching support 2.0.1?